ALSchwalm/pISO #16: Noob Issue with creating working image

jtoland, 2018-05-07:
Hello,

I am having an issue trying to get it to boot on both my Pi Zero and Pi Zero W.

I used a centos box with docker to build the sdcard.img then used win32diskimage to copy it to the SD card.

Can you confirm if this needs to be build on the PI or not?

Also would be nice if you just provided a blank image for us noobs out there.

Kind Regards,
Jason

ALSchwalm, 2018-05-07:
The build should not be run on the pi. It sounds like you did the right thing (and if you got the sdcard.img, then things should be working). I'm not familiar with win32diskimage, I plan on testing some windows programs for writing the image to a card, but it is not a priority at the moment.

There are actually automated builds the project, you can access them by going here: https://gitlab.com/ALSchwalm/pISO/commits/master

And then clicking on one of the green checkmarks. On the page that loads, there will be a small cloud with a down arrow on the right hand side. If you click that, there will be an option for "Download make-image artifacts". Clicking that will download  zip file that contains a "sdcard.img.tar.gz" which can be extracted and written to the sd card.

There are tagged releases here: https://github.com/ALSchwalm/pISO/releases
But they are somewhat out of date. I will probably tag a new release soon.

ALSchwalm, 2018-09-30:
For setup and usage instructions, please see [there wiki](https://github.com/ALSchwalm/pISO/wiki). I'm going to close this issue, but if you are still having problems, feel free to re-open it.

closed 2018-09-30 by ALSchwalm