ALSchwalm/pISO #25: Ensure that a sd card can be flashed multiple times

ALSchwalm, 2018-09-05:
Currently, the flash procedure can cause strange things to happen if a drive has already been flashed once. We should probably prevent that.

BigTexun, 2018-10-08:
I found this the hard way...  IF the filesystem is something Windows supports, etcher works fine.  If the filesytstem is a previous RP filesystem, etcher will corrupt the partition table.  No Windows based tool can correct this condition, the partition causes partition tools to lock up in Windows.  I was able to resolve this by using GParted.iso.  GParted boots into a linux kernel and has no trouble working with the partition table that locks up Windows.

Windows CAN delete Linux partitions in general, so I'm not sure why Etcher has a problem.  Etcher reports that it is using the diskpart command line, but I have not dug into what it is doing wrong.

Running diskpart, selecting the drive and issuing the clean command generates a "cannot find file" error, but  successfully wipes out the linux partitions, allowing etcher to run successfully.  It seems Etcher is running the /wrong/ diskpart command line for a flash devices with linux partitions.

BigTexun, 2018-10-08:
If I run etcher first, diskpart's "clean" function cannot work.  You MUST run diskpart's clean function before running etcher.  Once you have corrupted the partition table with etcher, you must be running linux to repair/clean the drive...  At least that is what I found.

I spent time resolving this last weekend, and today I was able to recreate the problem.  So even if I have some details wrong, I can recreate the issue, the corruption, and the two work-arounds I have tested.