ALSchwalm/pISO #30: Naming Drives

clreinki, 2018-09-29:
I really wish we could name each drive - even if it's limited to like 8 characters.  Something to better remember which drive is which.

kdallasd, 2018-10-01:
Something as simple as a text file in the root of each partition (like name.txt) that could be read (like how the ISOs currently are) would be nice. Actually inputting one via the piso would be a pain

ALSchwalm, 2018-10-02:
The trouble with using a text file on each virtual drive is that we would need to mount the drive 'internally' so the pISO can read it before we 'export' the drive so the host computer can see it. This would slow things down and potentially create issues when the user is trying to boot from an ISO.

One possible solution here is to add a section to the `piso.config` file that allows the user to provide a different name. Something like this might work:

```
[[drive]]
name="Drive1"
rename="My Drive Name"
```

This could also be useful if I end up adding other drive-specific options.

clreinki, 2018-10-02:
I like your proposed idea, Adam.