ALSchwalm/pISO #32: Trouble Mounting ISOS

codybuell, 2018-10-01:
I'm able to create and use drives just fine, however when trying to mount ISO's I get the error:

> The disk you inserted was not readable by this computer.

Working on OSX Mojave.  Have tried mounting ISOs with from EXFAT and FAT32 drives.  Believe I'm working with pISO v1.0.1.

ALSchwalm, 2018-10-02:
I am not sure why you're experiencing this issue. I have not updated to Mojave yet, but I will do so soon and see if I can recreate the problem. Thanks fo the feedback! I will get the issue resolved as soon as possible.

compukidmike, 2018-10-02:
I'm having the same issue on macOS Sierra (10.12.6), using piso v1.0.2

yllwfsh, 2018-10-07:
@codybuell What kind of file system is on the .ISO file? Can macOS mount anything from it when double clicking the iso file? I noticed that macOS is generally unable to mount Linux based .iso files, so this would be expected behaviour...

codybuell, 2018-10-07:
@yllwfsh They are linux iso's.  It was my understanding that the piso does not try to mount the iso on the host OS's filesystem but rather mounts itself internally and presents it to the host as a virtual CD drive.

yllwfsh, 2018-10-07:
@codybuell I'm not exactly sure how pISO mounts the .iso files, maybe @ALSchwalm can elaborate. I do know that under Linux (Debian/Ubuntu) all the ISO's mount fine, and PC's can also boot from them.

ALSchwalm, 2018-10-08:
Sorry for the slow start on this. I have confirmed the issue, though I'm not really sure what's causing it. It is next up on the list for me to work on after #35.

joedemo42, 2018-10-10:
FWIW I verified this issue with an ISO directly from Microsoft: Win10_1803_German_x64.iso - also this happens with FreeBSD-11.2-RELEASE-amd64-bootonly.iso or really any other ISO. As for the Win10 ISO I'm able to mount it just fine on macOS 10.13.6

ALSchwalm, 2018-10-16:
I would like to track this issue down next for a '1.3.0' release, but I think I am going to need a little help from someone. My MacBook is USB-C only, which is making it difficult to debug this kind of issue. If anyone has a MacBook with a standard USB port (that is, one where you can plug the pISO in without an adapter of any kind), and is willing to help debug, please let me know. You will also need to be running High Sierra or newer for this debugging.

Thanks in advance to any volunteers!

joedemo42, 2018-10-16:
I could do that. I‘ve got a 2012 MacBook Air running High Sierra.

ALSchwalm, 2018-10-16:
@joedemo42 Great! Here is the procedure:

1. Install [Wireshark](https://www.wireshark.org/)
2. From the terminal, run 'sudo ifconfig XHC20 up' (this enables capturing the USB traffic)
3. Open wireshark. You should see an 'XHC20' interface
4. Double click this interface to start the capture.
5. Insert the pISO and select an ISO
6. When the error message is displayed saying the disk is not readable by the computer, press the red stop button to stop the capture.
7. 'File->save as' and save the capture somewhere
8. Send the capture to 'usb@piso.support'

Thanks for your help! And please let me know if you have any questions or run in to any issues.

yllwfsh, 2018-10-16:
@ALSchwalm I'll also chip in with some packet captures. Sending you an email shortly.

joedemo42, 2018-10-16:
pcapfile was sent.

ALSchwalm, 2018-10-17:
Big thanks to everyone who helped out with the captures! I'm not exactly sure what this issue is here (though I suspect is has something to do with the MODE SENSE implementation). However, Wireshark apparently cannot correctly parse USB packets captured on a Mac, so I'll need to fix that first before I can really diagnose the issue here.

yllwfsh, 2018-10-17:
@ALSchwalm Maybe the issue is with wireshark? I'll try a Nightly build tonight (
https://www.wireshark.org/download/automated/osx/Wireshark%202.9.0-2227-g3d6bf1fe%20Intel%2064.dmg) Maybe that helps. What kind of packets (e.g. your packet filter) are you looking for? Then I can check if it's working properly or not...

ALSchwalm, 2018-10-22:
Just to keep people in the loop on this. I have not tracked down the exact issue, but I have patched wireshark so it can read USB captures made on MacOS correctly. For anyone interested I have attached the patch (it applies to wireshark as of wireshark/wireshark@7ca5dba864391807c50b8a843886b1131563f1fb): [wireshark.patch](https://github.com/ALSchwalm/pISO/files/2503744/wireshark.patch.txt)

I should be able to track things down now. Currently, it looks like the problem is that MacOS is issuing CD READ instead of standard READ commands, but it will take a bit more digging to be sure.

ALSchwalm, 2018-10-30:
This should be fixed in the latest v1.3.0-beta.1 release.

yllwfsh, 2018-11-04:
Some ISO's will load correctly under macOS now. Some don't, but I suspect that's not pISO's fault, but more because they are Linux/Unix/BSD specific iso's, e.g.:

Working:
Win10_1803_English_x64.iso
proxmox-ve_5.1-3.iso

Not Working: (The actual .iso file doesn't load on macOS either when double clicking, 'no mountable filesystems')
FreeNAS-11.2-BETA3.iso
gparted-live-0.32.iso
pop-os_18.04_amd64_nvidia_36.iso
xcp-ng-7.5.0-2.iso
elementaryos-5.0-beta2.20180920.iso

Is there any trick that you could think of with pISO to be able to mount these under macOS? Or is this simply a limitation of macOS itself?

commit ec852b18 (2018-10-24): Fix bug where discs would not mount is MacOS (fixes #32)

This was caused by MacOS issuing a READ DVD STRUCTURE command to determine
the type of disc being read. When we failed to respond to this, the OS
starts to attempt to read with CD READ requests instead of standard READ
requests. We don't support CD READ currently. Now we respond to the first
request correctly, so MacOS does normal READs and things work correctly.
commit d79209a9 (2018-10-24): Fix bug where discs would not mount is MacOS (fixes #32)

This was caused by MacOS issuing a READ DVD STRUCTURE command to determine
the type of disc being read. When we failed to respond to this, the OS
starts to attempt to read with CD READ requests instead of standard READ
requests. We don't support CD READ currently. Now we respond to the first
request correctly, so MacOS does normal READs and things work correctly.