ALSchwalm/pISO #33: Not able to boot from ISO files

lorddaveron, 2018-10-01:
I've created the ISOS folder and copied over a Windows 8 iso, Ubuntu Desktop iso and Ubuntu Server ISO.  I've followed the guide and activated the iso and when plugged into a running windows 10 OS I see the CD rom mount and the files come up.  But if I try to boot from PISO,  the computer doesn't see the image as a boot up option.  I've tried two different computers and several different BIOS settings.  I don't have a PISO error dump since there was no error.  It just won't recognize as a bootable device. I set the drive up as fat when I first provisioned it.  Shouldn't the PISO support this?

ALSchwalm, 2018-10-02:
I suspect the issue you're having is related to boot timing. There is a race when the computer starts up, if the BIOS polls USB devices before the pISO has started, then it will miss the virtual CD drive that the pISO creates and not show it as a boot option. There are two ways to get around this. The easiest is just to leave the pISO plugged in and soft-reset the machine (typically via something like ctrl-alt-delete from your BIOS screen). On most machines, this will leave the power to USB peripherals active but start the startup sequence again. On this second bootup, the pISO is already started, so it will be read by the BIOS and the boot option will be displayed.

The other option is to power the pISO via the micro usb ports. If you power the pISO via, for example, a microUSB phone charger, and plug it in to the machine, then it will already be booted when the machine checks for USB devices.

Please let me know if doing either of those resolves your issue or if you have any other questions.

ukscone, 2018-10-02:
>The other option is to power the pISO via the micro usb ports. If you power the pISO via, for example, a >microUSB phone charger, and plug it in to the machine, then it will already be booted when the machine >checks for USB devices.

I have been asking around about this as it seems like it could be slightly dangerous to the pizero or the computers usb slot as there would be power coming to the pi from two sources. I haven't tried it as i'm a teeny bit "scared" but if you say it'll be fine then i'll give it a go

ALSchwalm, 2018-10-02:
@ukscone You are correct, I asked @james-tate (the hardware designer for the pISO) and he confirmed it was not a good idea. I have edited my answer above accordingly.

lorddaveron, 2018-10-02:
I've tried powering the pISO with a USB charger and then booting the PC,
I've also tried setting up the .iso while the machine was on,  seeing the
drive pop up in windows and rebooting the machine.  The screen on the pISO
shows that the image is mounted,  but the bios does not see any bootable
device.  Powered versus un-powered doesn't seem to make a difference.

james-tate, 2018-10-03:
@lorddaveron Can you let me know the type of machine are you using so I can do some research on this?

ALSchwalm, 2018-10-03:
@lorddaveron Is it possible for you to confirm that the device supports booting from a normal USB CD drive? This is essentially what the pISO is emulating, so if it is not supported by your device, that would cause problems.

lorddaveron, 2018-10-03:
I've tried on a couple of different machines.

Dell Latitude E7450
Dell Optiplex 990
Custom Desktop with MSI Z87-G45 Motherboard

jact10, 2018-10-03:
having almost the same problem, but i can boot with linux isos stop in the first step saying that the cd is damage or currupt data, i try with 5 diferent micro sd card and different ISO image and having the same problem

GunnarSpawn, 2018-10-06:
I'm seemingly getting stuck for every linux iso I've used 32 bit or 64x on the "Failed to detect or get files from CD/Rom". 
Verifying the device works fine but then trying to get the content from the cd/rom fails everytime. When I boot from the linux cd gadget it seems to fail to mount any drive that either the piso or machine could use as a valid hard disc for installation. 

I've tried Centos and a ParrotOS Debian based iso and no go. I also cannot mount the cd roms dir from shell to a folder to try to load it from the installer.

GunnarSpawn, 2018-10-06:
Update:
What I have found works to bypass this issue is. Instead of loading the iso's from the ISOS folder withing the Virtual drives the Piso creates. I have found instead using the virtual drive and writing the operating system ISO to the virtual drive work better then the ISOS folder for the moment. Not too big a deal for me, so basically if you created several virtual drives then wrote ISOS to each of them, thats a compromise for now. The issue I was ultimately having was linux ISOS would fail to fully load files from the iso and upon boot loads a bunch of failed to mount errors. Using it this way you still may notice mount errors however you can load into live mode from your Virtual disc partitioned linux iso installer as well the installer just fine oddly enough. 

#NOTE# Picture 1 is an example of my virtual drive 2 being formatted to run as a ParrotOS debian ISO installer usb /drive. Drive1 using ISOS folder doesn't load as well as this method for some reason.

#Picture 1#
[image]

#Note# Picture 2 is an example during installation of the partition manager to just show what I was working with and how it could potentially show up as being mounted.
#Picture 2#
[image]

#Note# Picture 3 is an example of live image being ran from the virtual drive successfully vs Drive1 ISOS method.
[image]

Hope this helps anyone for the time being. I used win32discimager for the iso - virtual drive and I had it formatted under fat32.

Drive1 even being under fat32 wasn't able to load fully / properly from "Cd/rom" during installer and would fail to continue.

ALSchwalm, 2018-10-08:
@GunnarSpawn I believe the issue you and @jact10 are running in to should be fixed during the next release (at least for linux isos). I'm currently holding off on doing that release until I can get a fix in for #35 (or if I determine that will take longer than I thought). Hopefully I'll get it out in the next few days, though.

GunnarSpawn, 2018-10-08:
I can try to get you a better screenshot / log from the PISO if you like. It seemingly has troubles mounting directories / drives so I presume its also unable to fetch the files from the appropriate virtual drives iso / currently mounted one. Thank you for all your work with this @ALSchwalm I've been having an absolute blast messing around with it haha.

Even doing it this way for the moment is fine I can just use several virtual drives formatted correctly to the distro / os used and just write the isos to them using win32discimager / akin. Then it seems to be a little better when selecting that drive and using that to boot per my example the other day. 

I've basically got 3 drives at the moment 1 for isos / usb storage, second for windows ISO testing which I also encountered the driver error (using the ISOS folder method), and thirdly the one from my example; a virtual drive with iso installed onto it versus housed within a ISOS folder.

I've been absolutely loving the copying speeds from wifi versus physical so if anyone is having troubles with it taking a long / obscene amount of time. I shaved a few minutes using the wireless AP and ftp'ing to it.

teemusaila, 2018-10-10:
Booting does not work either with Lenovo W510. Tried with both methods, ISO and bootable USB drive.
Bios recognises the drive and starts to boot, but after a while it crashes without any message and boots from local HDD. Very same ISO's are bootable with Yumi on regular USB drive.

ALSchwalm, 2018-10-10:
This issues many users in this thread are experiencing may be fixed with the latest beta: [available here](https://github.com/ALSchwalm/pISO/releases/tag/v1.1.0-beta.1)

jact10, 2018-10-10:
My problem was solved with the BETA version! Thanks ALSchwalm!!!!!!

jact10, 2018-10-10:
I tryed with Windows XP/7/8/10 and works good, this night i will test it out with KALI and ubuntu and i Will let you know

GunnarSpawn, 2018-10-10:
@jact10 I had Parrotos Debian pentesting OS work I suspect the Kali one should work similarly. If you install the OS to the virtual drive you can even run it in live mode versus just use installer.

jact10, 2018-10-11:
Linux Woked perfectly fine.!!! all my problems has been resolved!

GunnarSpawn, 2018-10-11:
same, thanks for all your hard work! @ALSchwalm

lorddaveron, 2018-10-11:
I ended up rebuilding the image on the SD card with the new beta version
and so far it is working better.  I was able to see the image show up as an
option on one of my machines.  Thank you for the fixes.

ALSchwalm, 2018-10-16:
It sounds like this has been resolved. If anyone is still having issues, please ensure you are running a version of pISO OS 1.1.0 or newer.

closed 2018-10-16 by ALSchwalm