ALSchwalm/pISO #35: Trouble with Windows ISOs

ALSchwalm, 2018-10-03:
Some users are reporting issues with windows isos only shows a single text file when mounted with the pISO. I need test this out.

jact10, 2018-10-03:
I have same issue

jact10, 2018-10-03:
UDF file system iso image; thats the problem!  ALSchwalm please check!

jact10, 2018-10-03:
This disc contains a "UDF" file system and requires an operating system that supports the ISO-13346 "UDF" file system specification.

jact10, 2018-10-03:
maybe this work https://ubuntuforums.org/showthread.php?t=1751442

sudo mount -t udf /dev/dvd /mnt

jact10, 2018-10-03:
its going to be a little complicate, because you have to detect or the person manually have to choice what type of ISO is, like does YUMI

robiwisc, 2018-10-04:
I get the same thing on various Windows based ISOs. 

> This disc contains a "UDF" file system and requires an operating system
that supports the ISO-13346 "UDF" file system specification.

jact10, 2018-10-04:
And when I try to boot the iso, ask me for drivers

ALSchwalm, 2018-10-04:
@jact10 I have had issues with that when I am not booting EFI (which fails for some reason on certain machines). But if you can boot the virtual cd EFI, that will probably resolve your issue.

ALSchwalm, 2018-10-16:
If anyone is still having trouble with this, please let me know. The issue should be resolved in any release 1.1.0 or newer.

closed 2018-10-16 by ALSchwalm