ALSchwalm/pISO #37: Password protecting volumes

sauerjc65, 2018-10-03:
Is it possible to password protect the volumes on the pSIO? I’m browsing the documents but could not find anything.   Also would it support an external hard drive attached to the piZero?

ALSchwalm, 2018-10-03:
Unfortunately, we don't have plans to support password protecting volumes. This is mainly because there isn't an obvious way to have the user actually enter a password on the pISO (other than over bluetooth for Pi ZeroWs). Of course, normal encryption methods like BitLocker or LUKS will work with pISO virtual drives.

There is also no way to attach an external hard drive to the Pi Zero. This is because the USB controller on the Pi cannot act as both a USB client (for the connection to the PC) and a USB host (for the connection to the hard drive) at the same time.

I'm going to go ahead and close this ticket, but please feel free to open it again if you still have questions.

closed 2018-10-03 by ALSchwalm

sauerjc65, 2018-10-04:
I wish I knew all of this prior to purchasing. Basically,  this is just a fancy, large,  usb drive with a lot of unused computing power.

GunnarSpawn, 2018-10-13:
It said in the kickstarter page as well the comments during it's backing period what it could or couldn't do. People asked tons of questions and I asked actually the exact same question haha so I feel you.

 I love it personally  though it's a network capable usb or installer other times for me at least among other things. I've been running live images to test and running them persistent to a virtual drive to keep multiple going.

It's a limitation for the pi acting as a usb host or usb hid device same time so it's not exactly @ALSchwalm's fault he worked with what he had :p.