ALSchwalm/pISO #38: Can't boot USB bootable enabled Pi3B/3B+ using pISO

ukscone, 2018-10-03:
Only done cursory tests so far but not been able to use a pISO to boot an USB boot enabled Raspberry Pi. Only tried with a drive with IchigoJam BASIC4Rpi so far as it should have been the thing that would cause the least amount of problems as it's just files in a FAT32 partition.

Has anyone had any success booting a pi using pISO?

mikerr, 2018-10-06:
I had the same problem, it seems by default the pi doesn't wait long enough.

I've just successfully booted my pi3 from pISO  though:

The fix is to insert an sd card with only bootcode.bin and an empty file called TIMEOUT on it.
That adds 5 seconds to the wait for USB enumeration, which is enough for pISO to initialize.

I think there may be a config.txt flag to make that permanent in the OTP so you won't need to use an SD card again.

mikerr, 2018-10-06:
Just to add that was actually a Pi3B+ 
the same doesn't appear to work on my Pi3B despite it being USB-boot enabled ?!