ALSchwalm/pISO #42: Trouble copying files larger than 200MB over SMB

ALSchwalm, 2018-10-07:
A user has reported difficulty copying larger files (>200MB) via SMB.

joedemo42, 2018-10-10:
Can confirm. SMB advertises around 200MB of free space regardless of free space and SD card size (64 GB in my case, 5 GB used). Also, it would probably be better to serve each drive as an individual share - not all together as folders inside "mount" - as that would allow your SMB daemon to correctly report the free space for each individual drive.

ajanulis, 2018-10-10:
Same there - even 80MB is not possible. Commenting there just to be informed about activities

ALSchwalm, 2018-10-15:
This should be fixed in the latest beta: https://github.com/ALSchwalm/pISO/releases/tag/v1.2.0-beta.1

ajanulis, 2018-10-16:
Unfortunately, on OSx it doesn't work. See attached screenshots - even
Samba rejects to work.

ALSchwalm, 2018-10-16:
@ajanulis Your file did not attach correctly. I have tested this on MacOS Sierra and it is working correctly. Are you authenticating as the 'piso' user (or whatever user you have configured in your `piso.config`)?

ajanulis, 2018-10-16:
@adamschwalm@gmail.com <adamschwalm@gmail.com> are you connecting from OSx
via wifi? It works correctly if I just plug it into USB, but over wifi it
rejects on samba, o ftp it mounts as read only. And yes, I am using correct
usearname and password - otherwise I will not be able to connect to if with
ftp. This is why I've attached these screenshots

joedemo42, 2018-10-16:
Can confirm it works as expected. ~500MB successfully transferred. Thanks for the quick fix.

ajanulis, 2018-10-16:
@joedemo42 are you using Win or OSx? Wifi of USB?

joedemo42, 2018-10-16:
@ajanulis Wifi/SMB obviously. Also, MacOS X 10.13.6.

ajanulis, 2018-10-16:
@joedemo42 - hmm, same 10.13.6 here. How you power it? Maybe I got all this due to powering it from USB power bank? Have no idea where to start digging

joedemo42, 2018-10-16:
@ajanulis I just plugged it right into my 2012 MacBook Air. No additional power. Let me know if you need me to try something. Maybe I can help you!

ALSchwalm, 2018-10-17:
I just tested the latest beta on my 2017 MacBook Pro (also 10.13.6) and it seems to be working. Just to test, I also tried it while powered from a phone charger and had no issue.

ajanulis, 2018-10-17:
thanks all, tested with another MacBook and everything works as expected!
Sorry for fake warning - looks like my OSx is somehow screwed up :( time
for fresh installation. By the way, @adamschwalm@gmail.com
<adamschwalm@gmail.com> , as you are saving some customers settings in a
file, it might be good to save OLED orientation as well - it's not a big
deal, of course, rotate it each time I power it, but... just a nice feature.

Thank you once again and happy hacking!

p.s. I also had an idea to use it as flash drive for music in a car (with 2
different playlists on different virtual Drives - one for my, one for my
girlfriend), but that BT/USB player/amplifier can't access drives as then I
plug it in, pISO stays in "loading" state all the time.

Aidas

commit a852ddf0 (2018-10-14): Use separate smb shares for each partition (fixes #42)