ALSchwalm/pISO #44: add filesystem type to virtual drive listing

dannen, 2018-10-08:
I'd like to see the format of each virtual drive listed on the right side of the drive listing in the main menu.

I.E.

Drive0 (NTFS)
Drive1 (EFS2)
Drive2 (EXFAT)

There is no way currently to remember which format I assigned to which virtual drive.

GunnarSpawn, 2018-10-09:
You can rename the drives potentially if that could help you out a bit. I think I was able to do it within windows?

ALSchwalm, 2018-10-09:
I had considered this, but I do think the renaming feature that will be in the next release would be a better way to do this. Each virtual drive could actually have multiple partitions (if the users chooses to repartition the drive that way), so there is no good way for the pISO to list a specific type for every drive.

ALSchwalm, 2018-10-10:
Renaming is now supported in the latest beta release [available here](https://github.com/ALSchwalm/pISO/releases/tag/v1.1.0-beta.1), so you can rename your ext3 drive to 'Linux Drive', for example. Does that resolve this issue?

dannen, 2018-10-15:
yes, that will help! I'll download the beta and give it a try.