ALSchwalm/pISO #45: Adding first [[wifi.client]] entry works, others fail

BigTexun, 2018-10-08:
When I add a total of three wifi.client entries in piso.config, all three show up in the wifi menu.  However only the first entry works.  Selecting the second or third client entry results in a connection using the first entry.  If the first entry is for an AP that is out of range, the connection fails with a error indicating it was unable to get an IP address.  So selecting other wifi.client entries seems to do nothing, it only attempts connections with the first entry.

2600Sam, 2018-10-16:
I have the same problem.

ALSchwalm, 2018-10-16:
Thanks for the verification. I'm not sure exactly what's causing this, but I'll try to track in down in the next week or so.