ALSchwalm/pISO #47: BluRay ISO not working

hfjensen, 2018-10-10:
I copied the ISO from a ripped BluRay and a ripped DVD to the ISOS folder, but after mounting either ISO the computer where I plug in the pISO does not recognize the image as a DVD or a BluRay respectively.
I can however mount the standard disc, and with a virtual DVD/BluRay SW mount the ISOs from the ISOS folder, and then play them.
The pISO mounted BluRay ISO is not recognized at all and the pISO mounted DVD is reported a 'corrupt'

GunnarSpawn, 2018-10-10:
I think it's meant for more operating system installers. For what you're doing why don't you just not copy it to an isos folder, rather just the root of the drive and mount on the machine whenever you have the drive pop up?

ALSchwalm, 2018-10-10:
I have not done any testing with BluRays, but this issue may be fixed in the latest beta (which fixes a lot of things with larger isos): [available here](https://github.com/ALSchwalm/pISO/releases/tag/v1.1.0-beta.1)

hfjensen, 2018-10-19:
You were right, I'm now using the latest update v1.2.0-beta.1, and DVD and BluRay ISO's work fine.

closed 2018-10-19 by hfjensen