ALSchwalm/pISO #49: Upload data via WiFi while Drive is connected ends up in limbo

joedemo42, 2018-10-10:
Testsystem: macOS 10.13.6. 
Let Drive2 be single partition ExFAT. 
Have Drive2 connected and copy data (data1) to Drive2. Verify data1 is there using Finder. 
Connect to pISO via SMB. Check mounted Server in Finder. See "Drive2 (partition 1)". 
Check contents of Drive2 (partition 1): No data1. 
Copy data (data2) over WiFi to Drive2 (partition 1). Verify data2 is there in mounted share.
Check Drive2 (connected via USB): no data2, just data1.
Disconnect everything, reconnect everything: still data1 only visible via USB, data2 only via WiFi.

ALSchwalm, 2018-10-15:
This should be fixed in the latest beta: https://github.com/ALSchwalm/pISO/releases/tag/v1.2.0-beta.1

joedemo42, 2018-10-16:
Can confirm it works as expected. Thanks for the quick fix.