ALSchwalm/pISO #55: Drives issue

Marnunz, 2018-10-15:
After a connection with usb-c (as shown in the attached picture) the pISO doesn't recognize drives anymore. I can create every drives i want,everythings seems alright, but my iMac and also Win 10 O doesn't load anything.

[image]

ALSchwalm, 2018-10-15:
The Raspberry Pi Zero has MicroUSB ports not USB-C ports, are you sure you're using a usb-c cable? I wouldn't expect that to even power on.

Marnunz, 2018-10-15:
[image]
[image]

ALSchwalm, 2018-10-17:
So can you confirm that the cable you were initially using (the black one) was USB-C and not micro usb? And you were previously able to view the virtual drives you create?