ALSchwalm/pISO #57: Allow size_step and default_size to be floats

ALSchwalm, 2018-10-16:
Users with larger SD cards should be able to create arbitrarily small drives.

ALSchwalm, 2018-10-16:
Really, there should probably also be an option to allow the user to specify the step size in GB (also float)

BigTexun, 2018-10-21:
I was going to post a request for better sizing flexibility.  For example, we have a network install for windows 10 that takes 384k, so on a 128g flash module, the minimum 5g size is 90% wasted.  being able to dial in any size is useful as we often know exactly how much space we need for a given image, so it would save a lot of space.  So this would be a /great/ feature.  128g is $25 on Amazon, only a matter of time before the 256g are $25...  not wasting all of that space would be great.

TheOneValen, 2018-10-24:
@BigTexun 

In the meantime: You can configure the step size and make the drives smaller. Also due to automatic over provisioning your drive is not actually using the 5G. Just the space you use.