ALSchwalm/pISO #66: Locking up upon creating or deleting a drive

tinjaw, 2018-10-19:
I just flashed my sd and put it in and it booted up just fine. I can create a drive, but the onscreen menu locks up. I can pull the pISO out and reinsert and it boots up fine. The drive is there and usable. But if I delete it, the same thing happens. It freezes up.

ALSchwalm, 2018-10-22:
Hi tinjaw, thanks for the report. Can you please create a file called 'piso_debug_log' on your SD card. Then insert it in to the pISO and plug the pISO in to a computer. When you create a drive (and it locks up), unplug the pISO and the SD card. There should be a file called 'piso.log' on the SD card. Upload that here and we should be able to help debug things.

tinjaw, 2018-10-30:
[piso.log](https://github.com/ALSchwalm/pISO/files/2527588/piso.log)
 I finally found some time to try it. Here is the log file.

For this test, I downloaded a fresh img from the releases. I flashed a brand new image to a 64 GB SD. I did not plug it into a USB port. To make sure it had nothing to do with my computer, I plugged power into the pISO directly. Created a disk and it froze.