ALSchwalm/pISO #75: Support USB mass storage for virtual drive storage

piit79, 2018-10-25:
It would be great if we could use external storage, e.g. a high-speed USB stick to store virtual drives in addition to/instead of the SD card. SD cards are usually quite slow and this could possibly provide better performance.

ALSchwalm, 2018-10-25:
Unfortunately, it is not possible for the Raspberry Pi Zero to act as both a usb host and peripheral at the same time, so this is not workable. However, future versions of the Pi Zero might add this capability. For now, though, I'm going to close this issue, but thanks for the feedback!

closed 2018-10-25 by ALSchwalm

piit79, 2018-10-25:
Oh, didn't know that. That's a shame! I guess I'll invest in a much faster SD card for now :)

Just wanted to add that pISO is great - immediately used the ISO CD emulation feature a few times after getting mine. So nice not having to find an empty stick and write the ISO files.