ALSchwalm/pISO #78: Feature request for the distant future - Encryption

simontravaglia, 2018-10-29:
Just thinking about it, it would be nice to have the ability to have an encrypted drive.
Maybe when you mount the drive you're prompted for a password:
 Up/Down    cycles through the alphabet for each character
 Select          Selects the letter and advances one character (which scrolls round on itself)
 At the end/beginning of the password there's a <BACK> so you can abort the mount.

jgoergen, 2019-07-11:
I formatted a drive on it as a bitlocker encrypted volume on windows 10 and it's been working great that way as well.

simontravaglia, 2019-07-11:
Yes, I think I'll end up splitting the device into encrypted and unencrypted  volumes as there's some stuff I want available to all platforms (i.e. media) and personal stuff I want protected.   Ta