ALSchwalm/pISO #81: Card speed issues

sunbit, 2018-10-30:
I have speed issues copying to/from a pISO drive:

- SD card is a Samsung EVO Plus 128 Gb, class 10.
- Copy tests done with a single 4.57 Gb file.
- v1.2 firmware

Testing with the raw sdcard, 1 partition of 128GB ExFat:
- Write speed: 77 Mb/s
- Read speed: 97 Mb/s

Testing with a single drive created in pISO, 100% of capacity
- Write speed: 5.5 Mb/s
- Read speed: 14 Mb/s

I don't now if this can be caused by the raspberry pi or the pISO, but as you can see I have a 15x difference between copying files to/from the raw sdcard, vs copying files from a drive defined using pISO with the same card.

Any idea?

Thank you

appcoders, 2018-11-17:
Yes would be interesting if its a Raspi hardware issue...

xela92, 2018-11-27:
I'm facing this issue too, I have Samsung EVO Plus 128GB too, and I'm doing less then 5 MB/s writing.

Willena, 2018-12-23:
I've also face this issue. I looked through the internet and here is what I found 

http://www.pidramble.com/wiki/benchmarks/microsd-cards

There is nothing about the Pi Zero itself but I think it's similar to other Pis.
Most interestingly this is the of the last sentence of the author : " (Note that most of the above benchmarks, when run on a USB 3.0 card reader on my MacBook Air, show 5, 10, or 15 times greater performance in that environment)"

extremeshok, 2019-04-02:
Create the volume as ext3 and test, the nfs/fat32 is extremely slow.

im getting a 3x boost with ext3

Willena, 2020-02-16:
I've been trying things today and found out that things can somehow probably be improved.

Before anything, here are the measurements made using the pISO:
* Read : ~17MB/s
* Write: ~5MB/s

I tested the read and write speed of my micro sdcard from a fresh install of raspbian, using the script in [this ](https://www.raspberrypi.org/forums/viewtopic.php?t=144493) thread.  The result I got were encouraging !
* Read : ~18MB/s
* Write: ~17MB/s

As suggested by the thread, I added the `dtparam=sd_overclock=100` in my `/boot/config.txt`. The previous results has almost doubled:
* Read: ~38MB/s
* Write: ~32MB/s

This shows us that USB read write are being slowed by either LVM or the usb_gadget module (g_mass_storage). pISO is runnning on usb_gadget module of the kernel. pISO seams to be also using LVM for the drive management (please correct me if i'm wrong). 

Using only a bin file NTFS formated and usb mounted (`modprobe g_mass_storage file=/data/file.bin ro=0 stall=0`) I got:
* Read: 31Mb/s
* Write: 10-17Mb/s
These read write speed even if they are slower than what the raspberry pi can handle (see second sdcard test, with `dtparam=sd_overclock=100`) are more acceptable.  For the moment theses are just measurement and tests. It seams that the only thing left to test is LVM...

To be continued 😄