ALSchwalm/pISO #83: Drives not recognized by Windows 10 Pro

pmeyerdk, 2018-11-01:
Hi,

I have the following issues, when trying to connect pISO to my Windows 10 Pro PCs USB port.

1) When inserting pISO into my USB port, it seems to boot succesfully and show the menu.
But my Windows displays the following error message: "USB device not recognized".
2) When then trying to created drives on pISO, it seems to create and format succesfully.
But my Windows displays the following error message, when activating a drive in the pISO menu: "USB device not recognized".

As mentioned, I am running Windows 1 Pro.
My PC is a Dell XPS 15 9570, with and i7-8750H processor, 16 GB RAM and 512 GB.

I haven't had the chance to try pISO in another PC yet.
But wanted to report this, since I usually don't have issues connecting USB key to this laptop.

Regards,
Peter Meyer

ALSchwalm, 2018-11-02:
Hi @pmeyerdk what version of the pISO OS are you using?

pmeyerdk, 2018-11-02:
Hi Adam,

I am using version 1.3.0, more precisely the newest version I could find,
being version 1.3.0-beta.2.

/Peter

pmeyerdk, 2018-11-02:
I just tried to install v.1.2.0 instead of the newest one.
Though, this didn't make it work better, since I still get the same error message.

My SD card is a SanDisk Ultra 32GB HC I.
And I'm using Etcher to put the system on the SD card.

/Peter

ALSchwalm, 2018-11-03:
Thanks for helping debug this. Can you try connecting the raspberry Pi to you machine via a micro USB cable directly to the raspberry Pi? This will eliminate any potential hardware issue with the pISO USB header.

pmeyerdk, 2018-11-04:
No problem!

Connecting to my PC using a cable in one, actually both, USB ports of the
Pi, did not work any better.
One port had the same errors, and the other seemed to not have any
reactions on the PC at all.

Can it be pin connections not aligned completely?

It seems a little puzzling, since pISO seems to work while being used.
So it must be able to read the SD card, and get some power as well.

/Peter

ALSchwalm, 2018-11-07:
If the issue is still occurring when the USB connection is being made directly with the Raspberry Pi, it is probably not a problem with the pins. I suspect the issue is with the USB controller on the Pi (I have seen a couple of cases of this). Do you have another Raspberry Pi you could test with? If not, we can test the Pi if you have experience using a serial cable to connect to your Pi.

pmeyerdk, 2018-11-21:
Hi again,

Do you have any pointers, as to how I could continue testing this issue, to hopefully get it solved?

Thanks :-)

ALSchwalm, 2018-11-21:
Sorry for the slow reply on this. For your other Pi's, are any of them configured to use wifi and give you ssh access? If so, you may be able to just transfer the SD card to the Pi Zero you're using and get in to it that way (if it is a Pi Zero W).

pmeyerdk, 2018-12-30:
Hi again,

I got a friend, who's working with electronics and embedded software, to take a look at my pISO.
He took it to work, to check it with proper equipment and found the cause of my issue.

Pogo pin USB D+ and D- is shortwired.
So, apparently my issue is caused by an error in production.

My friend fixed my pISO for me.
But I haven't had a chance to check it myself yet though.
I wouldn't have had a change to find this our myself.

Best regards,
Peter Meyer