ALSchwalm/pISO #84: pISO Doesn't seem to boot

TheEagle13, 2018-11-08:
I installed my pISO, and flashed the SD card according to the instructions. I get the pISO loading screen, then it disappears and nothing comes up. If I hook up the Pi to HDMI, I just get a rainbow square on the TV. Any suggestions?

Loading...
[image]

After Loading...
[image]

HDMI Output...
[image]

TheEagle13, 2018-11-10:
@ALSchwalm I am waiting patiently for a response for suggestions.

@Thomas-Bastiaens 
This is not related, please don't hijack other's threads with irrelevant info. If you cannot find a relevant thread, make your own.
> I had first an error, but I didn't got an error file (piso_debug.tar), and now created a piso_debug_log that didn't work.
> So I reinstall the system and I had to boot it up for 3 times for it worked.
> And after a 10 days it stopped working. And I waited 5 min and it was still booting.
> And I reinstall, now it work. That its not smart if I save files on it.
> And so I have the same problems.
> try to boot it a couple of times and let know if it worked for you.
> 
> Sorry for my bad English!

ALSchwalm, 2018-11-12:
@TheEagle13 Sorry for the delayed response to this. Can you please a filed called 'piso_debug_log' on the root of your sd card. This should cause the pISO to create a file called 'piso.log' which may contain info helpful in diagnosing this.

khuzthul, 2018-11-12:
same issue here, I´ve tried with various sdcards and is always the same.
I have managed to launch piso two times, but it gets stuck when formatting the drives.
I attatch the debug logs i have, please let e know if there is something to do.
[piso_error.log](https://github.com/ALSchwalm/pISO/files/2571603/piso_error.log)
[piso_format_drive.log](https://github.com/ALSchwalm/pISO/files/2571604/piso_format_drive.log)
[piso_error_creating drives.log](https://github.com/ALSchwalm/pISO/files/2571605/piso_error_creating.drives.log)

[piso_format_drive.log](https://github.com/ALSchwalm/pISO/files/2571596/piso_format_drive.log)
[piso_error.log](https://github.com/ALSchwalm/pISO/files/2571598/piso_error.log)
[piso_error_creating drives.log](https://github.com/ALSchwalm/pISO/files/2571599/piso_error_creating.drives.log)

TheEagle13, 2018-11-12:
@ALSchwalm I don't have one. It never produces one. I should have thought to mention that in the initial post.
I'm not sure why people (like @khuzthul) keep hijacking threads with unrelated issues (probably because they don't know how github works ) but none of these other posters seem to have issues related to mine. I cannot get pISO to launch ever, no error logs ever produce.

khuzthul, 2018-11-12:
> @ALSchwalm I don't have one. It never produces one. I should have thought to mention that in the initial post.
> I'm not sure why people (like @khuzthul) keep hijacking threads with unrelated issues [...]

@TheEagle13 sorry to bother you, i have replied only because i have the same issue. the only difference is that i have the error logs to show.

@ALSchwalm please let me know if i have to open another issue, or if you need more data.

TheEagle13, 2018-11-12:
@khuzthul Not to start an argument, but if you read, my issue is that I cannot get pISO to load at all, yours is that you have issues with formatting.

rlights, 2018-11-12:
Just got my pISOs and have the same issue. Loading and then nothing, Unfortunately I don't have a HDMI adaptor handy, but I am also not seeing any logs produced after inserting the trigger file.

@ALSchwalm I did notice [some pogo pins aren't seating properly](https://photos.app.goo.gl/zsmpyoUYcZwAoGg16); I doubt this would prevent the Pi booting by itself, but maybe?

ALSchwalm, 2018-11-12:
@TheEagle13 To be clear, you have created the 'piso_debug_log' file on your SD card and it still does not product a debug 'piso.log' log file?

TheEagle13, 2018-11-12:
My apologies, I misunderstood that as a statement as opposed to a request. I created the file, and the only thing created in the piso.log is as follows:

`Building display manager
Building USB gadget
Building pISO`

ALSchwalm, 2018-11-13:
@TheEagle13 That's very strange. I've gone ahead and made a build with some more logging that will hopefully give some idea of at least what's happening there. I'm assuming that some operation is hanging during that setup phase, but I'm not sure what it could be. When you get a change, please update to this and get the log. 
[update-more-logging.zip](https://github.com/ALSchwalm/pISO/files/2578571/update-more-logging.zip)

ALSchwalm, 2018-11-13:
@rlights How are you connecting the pISO? Could you try using a micro usb cable to the PI instead of the USB header on the pISO?

steve500, 2018-11-14:
Same exact issue with mine that I had received today. :( Tried all the same suggestions, mine will not spit out a log. With the same SD cards, I can run other Pi software releases without issues. I am using 64gb sandisk sdxc cards. This Pi Zero is brand new, clean contacts. The pISO just splashes the boot logo for a couple seconds and then nothing.

ArdjanB, 2018-11-14:
I had the same issue. Under 1.2.0 everything worked fine, but when I updated to 1.3.0 it seemed to be broken..
* When booting in 1.3.0, the first logo appeared, and it got black again. Normally the logo appears a second time, but now it stayed black.
* Downgraded to 1.2.0 again, but the same issue: 1 logo, then black.
* Tried to get a log file: under 1.2.0 and 1.3.0 that didn't work
* Tried the update-more-logging.zip, and that worked: the pIso booted (1.3.0), I could mount disks, etc. This time I got a log file:
[piso.log](https://github.com/ALSchwalm/pISO/files/2583035/piso.log)

steve500, 2018-11-15:
ArdjanB, I tried the update-more-logging.zip update files and still cannot get my piso board to boot. Did we get duds?

ArdjanB, 2018-11-15:
It seems to be solved. I can now boot normally in 1.3.0.

I can't exactly pinprick the reason, but I had some problems with copying the 3 update file to the MicroSD card. Sometimes the copying would not start at all, etc. But I just now tried to delete the three files first, before I copy the new files, and that worked every time, with both 1.3.0 and these update-more-logging files.
I would currently say it was an incorrect overwriting of (one of) the three update files. I just dragged and dropped the files under Windows 8.1 and Windows 10.

TheEagle13, 2018-12-05:
@ALSchwalm 
Sorry about the delay in getting back to you and testing, I was on vacation. Anyway, using the Update more logging that you sent, I was able to boot once, and (i think) create a drive of 95%, but it no longer boots after the drive was created. Here are the logs. Also as a side note when it was booted, every time I would press a button it would shift the screen to the left by about 10%.

```Building display manager
Building USB gadget
Building pISO
Constructing VolumeGroup
Building drives from VolumeGroup
Building NewDrive menu item
Building Stats menu item
Building WiFi menu item
Building Options menu item
Focusing
Shifting focus to window id=2
Configuring user
  Adding user
  Doing bindfs
Finishing pISO construction
Restoring State
Rendering pISO
Positioning windowid=1
Blitting to (0, 0, size=(0, 0) scroll_shift=0)
Positioning windowid=2
Blitting to (0, 0, size=(82, 9) scroll_shift=0)
Positioning windowid=3
Blitting to (0, 9, size=(55, 12) scroll_shift=0)
Positioning windowid=5
Blitting to (0, 21, size=(31, 9) scroll_shift=0)
Positioning windowid=6
Blitting to (0, 30, size=(47, 9) scroll_shift=0)
Positioning windowid=4
Blitting to (119, 0, size=(9, 40) scroll_shift=0)
Update display with bitmap: 128 by 64
Building controller
controller: UP released without press
controller: DOWN released without press
controller: SELECT released without press
```

TheEagle13, 2018-12-15:
@ALSchwalm Any update? Did I get a dud?

closed 2018-12-31 by TheEagle13

TheEagle13, 2019-02-26:
To add some info to those who may find their troubleshooting steps similar to mine....
The issues appear to be the cheap pogo pins not pushing all the way through. After much troubleshooting and lots of testing of the pins, I would always find 1 or 2 that didn't push all the way through the holes.

I went ahead and just soldered the darn things together, and I have no issues anymore.