ALSchwalm/pISO #88: error after deleting

khuzthul, 2018-11-16:
I have managed to create 3 partition on the sdcard with pISO but trying to delete one of them ends in an error and the pISO wont boot again.

this is the error i get
_

> Error while processing 'DeleteDrive("Drive2")': Error: Command "lvchange" (["-a", "n", "VolGroup00/Drive2"]) failed:   WARNING: /dev/VolGroup00/Drive1: Thin's thin-pool needs inspection.
> Error while processing 'DeleteDrive("Drive2")': Error: Command "lvchange" (["-a", "n", "VolGroup00/Drive2"]) failed:   WARNING: /dev/VolGroup00/Drive1: Thin's thin-pool needs inspection.
> Error while processing 'DeleteDrive("Drive2")': Error: Command "lvchange" (["-a", "n", "VolGroup00/Drive2"]) failed:   WARNING: /dev/VolGroup00/Drive1: Thin's thin-pool needs inspection.
> Error while processing 'DeleteDrive("Drive2")': Error: Command "lvchange" (["-a", "n", "VolGroup00/Drive2"]) failed:   WARNING: /dev/VolGroup00/Drive1: Thin's thin-pool needs inspection.
>   /dev/mapper/VolGroup00-Drive1p1: read failed after 0 of 4096 at 0: Input/output error
>   /dev/mapper/VolGroup00-Drive1p1: read failed after 0 of 4096 at 4096: Input/output error
>   WARNING: /dev/VolGroup00/Drive2: Thin's thin-pool needs inspection.
>   /dev/mapper/VolGroup00-Drive2p1: read failed after 0 of 4096 at 0: Input/output error
>   /dev/mapper/VolGroup00-Drive2p1: read failed after 0 of 4096 at 4096: Input/output error
>   WARNING: /dev/VolGroup00/Drive3: Thin's thin-pool needs inspection.
>   /dev/mapper/VolGroup00-Drive3p1: read failed after 0 of 4096 at 0: Input/output error
>   /dev/mapper/VolGroup00-Drive3p1: read failed after 0 of 4096 at 80508551168: Input/output error
>   /dev/mapper/VolGroup00-Drive3p1: read failed after 0 of 4096 at 80508608512: Input/output error
>   /dev/mapper/VolGroup00-Drive3p1: read failed after 0 of 4096 at 4096: Input/output error
>   Logical volume VolGroup00/Drive2 is used by another device.
> Error while processing 'FormatDrive': Error: Command "lvcreate" (["-V", "13418208256B", "-T", "VolGroup00/thinpool", "-n", "Drive4"]) failed:   WARNING: /dev/VolGroup00/Drive1: Thin's thin-pool needs inspection.
> Error while processing 'FormatDrive': Error: Command "lvcreate" (["-V", "13418208256B", "-T", "VolGroup00/thinpool", "-n", "Drive4"]) failed:   WARNING: /dev/VolGroup00/Drive1: Thin's thin-pool needs inspection.
> Error while processing 'FormatDrive': Error: Command "lvcreate" (["-V", "13418208256B", "-T", "VolGroup00/thinpool", "-n", "Drive4"]) failed:   WARNING: /dev/VolGroup00/Drive1: Thin's thin-pool needs inspection.
>   /dev/mapper/VolGroup00-Drive1p1: read failed after 0 of 4096 at 0: Input/output error
>   /dev/mapper/VolGroup00-Drive1p1: read failed after 0 of 4096 at 4096: Input/output error
>   WARNING: /dev/VolGroup00/Drive2: Thin's thin-pool needs inspection.
>   /dev/mapper/VolGroup00-Drive2p1: read failed after 0 of 4096 at 0: Input/output error
>   /dev/mapper/VolGroup00-Drive2p1: read failed after 0 of 4096 at 4096: Input/output error
>   WARNING: /dev/VolGroup00/Drive3: Thin's thin-pool needs inspection.
>   /dev/mapper/VolGroup00-Drive3p1: read failed after 0 of 4096 at 0: Input/output error
>   /dev/mapper/VolGroup00-Drive3p1: read failed after 0 of 4096 at 80508551168: Input/output error
>   /dev/mapper/VolGroup00-Drive3p1: read failed after 0 of 4096 at 80508608512: Input/output error
>   /dev/mapper/VolGroup00-Drive3p1: read failed after 0 of 4096 at 4096: Input/output error
> Doing actions: [FormatDrive]
> Error while processing 'FormatDrive': Error: Command "lvcreate" (["-V", "13418208256B", "-T", "VolGroup00/thinpool", "-n", "Drive4"]) failed:   WARNING: /dev/VolGroup00/Drive1: Thin's thin-pool needs inspection.
> Error while processing 'FormatDrive': Error: Command "lvcreate" (["-V", "13418208256B", "-T", "VolGroup00/thinpool", "-n", "Drive4"]) failed:   WARNING: /dev/VolGroup00/Drive1: Thin's thin-pool needs inspection.
> Error while processing 'FormatDrive': Error: Command "lvcreate" (["-V", "13418208256B", "-T", "VolGroup00/thinpool", "-n", "Drive4"]) failed:   WARNING: /dev/VolGroup00/Drive1: Thin's thin-pool needs inspection.
>   /dev/mapper/VolGroup00-Drive1p1: read failed after 0 of 4096 at 0: Input/output error
>   /dev/mapper/VolGroup00-Drive1p1: read failed after 0 of 4096 at 4096: Input/output error
>   WARNING: /dev/VolGroup00/Drive2: Thin's thin-pool needs inspection.
>   /dev/mapper/VolGroup00-Drive2p1: read failed after 0 of 4096 at 0: Input/output error
>   /dev/mapper/VolGroup00-Drive2p1: read failed after 0 of 4096 at 4096: Input/output error
>   WARNING: /dev/VolGroup00/Drive3: Thin's thin-pool needs inspection.
>   /dev/mapper/VolGroup00-Drive3p1: read failed after 0 of 4096 at 0: Input/output error
>   /dev/mapper/VolGroup00-Drive3p1: read failed after 0 of 4096 at 80508551168: Input/output error
>   /dev/mapper/VolGroup00-Drive3p1: read failed after 0 of 4096 at 80508608512: Input/output error
>   /dev/mapper/VolGroup00-Drive3p1: read failed after 0 of 4096 at 4096: Input/output error

_and this is the complete log.
[piso.log](https://github.com/ALSchwalm/pISO/files/2590058/piso.log)

please help me. its been a nigthmare to work with this.

ALSchwalm, 2018-11-18:
Hi, I'm sorry to hear your having trouble. I've not seen something like this before, what kind of SD card are you using?

khuzthul, 2018-11-19:
hi again

i have tried with two different sdcards, and its always the same. first of all, when it works, it let me create some partitions, but when i try to reconnect it it allways collapse. in some cases, it doesnt boot, and in other cases, it wont format any other partitions, and then nothing happens again.

this is the last error i have.

_

> Error while processing 'FormatDrive': Error: Command "lvcreate" (["-V", "80509247488B", "-T", "VolGroup00/thinpool", "-n", "Drive2"]) failed:   WARNING: /dev/VolGroup00/Drive1: Thin's thin-pool needs inspection.
>   /dev/mapper/VolGroup00-Drive1p1: read failed after 0 of 4096 at 0: Input/output error
>   /dev/mapper/VolGroup00-Drive1p1: read failed after 0 of 4096 at 53673394176: Input/output error
>   /dev/mapper/VolGroup00-Drive1p1: read failed after 0 of 4096 at 53673451520: Input/output error
>   /dev/mapper/VolGroup00-Drive1p1: read failed after 0 of 4096 at 4096: Input/output error
>   device-mapper: message ioctl on  (254:2) failed: Operation not supported
>   Failed to process thin pool message "create_thin 2".
>   Failed to suspend VolGroup00/thinpool with queued messages.

_

ALSchwalm, 2018-11-21:
Do you have access to a linux or Mac machine? I would like to completely erase the contents of the SD card by writing all 0's to it (and I can provide the commands to run if you have access to one of those). That said, I don't know how this kind of issue could be affecting two different cards unless there is a problem with the Pi. Have you booted different Raspberry Pi operating systems on that Pi with those SD cards?

khuzthul, 2018-11-21:
hi again, i have access to linux and mac, but i have to say that i have tried this a couple of times and it always ends in error.
last time i managed to boot pISO and create 2 partitions, even i managed to copy isos to one of them, but only the first time, when i disconnect from the pc it doesnt work again.
after that i only get this error _Error: No such file or directory (os error 2)_ and pISO doesnt boot at all.
please send me your commands and i will try.

khuzthul, 2018-11-22:
hi again, i have been doing more test to my pISO.
i have tried this for three times and always is the same results.
1º managed to format in low level the sdcard and format it in FAT32
2º install the OS with ETCHER two times, because the first always fails
3º after install plug pISO in my laptop and wait to boot
4º create the first partition and wait for 10 minutes, nothing happens
5º unplug pISO and check log:
_> Doing actions: [FormatDrive]
> Error while processing 'FormatDrive': Error: Command "lvcreate" (["-V", "53672833024B", "-T", "VolGroup00/thinpool", "-n", "Drive1"]) failed:   device-mapper: reload ioctl on  (254:2) failed: Block device required
>   Failed to activate thin pool VolGroup00/thinpool._
6º plug in again and create again the same partition, this time goes well
7º unplug pISO again to see the log 
_> Doing actions: [CloseFormatMenu, CreateDrive(LogicalVolume { name: "Drive1", path: "/dev/VolGroup00/Drive1", size: 53674508288 })]
> Shifting focus to window id=2_
8º plug in again and then nothing more pISO dont work again.

please help me, i think my pISO is not working well. I have tried the Raspberry with raspbian and it works well with this sdcard.
is there any way to unplug pISO securely or something like that?
how can i create my partitions and have it work without losing it when i disconnect?

thank you.