ALSchwalm/pISO #89: pISO UI glitches around.

MithicalM, 2018-11-20:
What the title says.
Video: https://youtu.be/2XgyFz_1lqA

ALSchwalm, 2018-11-21:
Hi @MithicalM, I'm sorry your having issues with your pISO. Issues like this are typically caused by a poor connection between the pISO and the Raspberry Pi. Could you ensure all of the screws are tightly screwed on?

ArdjanB, 2018-11-26:
I am seeing the same things. I'm on 1.3.0

TheEagle13, 2019-01-01:
@MithicalM @ArdjanB @Thetmar 
I had the same issues, and it actually was a poor connection between the pins. I found that the included spacers were actually a hair too tall. after removing them and reinstalling (tighten so it's tight but not warped) and you should be good to go.

TheEagle13, 2019-02-26:
To add some info to those who may find their symptoms similar....
The issues appear to be the cheap pogo pins not pushing all the way through. After much troubleshooting and lots of testing of the pins, I would always find 1 or 2 that didn't push all the way through the holes.

I went ahead and just soldered the darn things together, and I have no issues anymore.

MithicalM, 2019-02-26:
Honestly. I have given up at this point. This thing is really badly made
and I've just had constant issues and no real help or feedback from the
devs on how to get it sorted. Just unacceptable really.

TheEagle13, 2019-02-26:
I agree, they farmed out production to the lowest bidder, and it shows (But it was a kickstarter, not a retail product). I'm not too sure what happened to the Dev team, they probably got tired of the complaints. But the concept, and at least for me the use cases, are really practical and useful. Much better than the Big Zalam Virtual CD Drive thing I've been carrying around for years. Once I did the in depth troubleshooting and discovered that soldering the pins was the fix, I haven't had an issue since. (And I had issues just as much as you did)