ALSchwalm/pISO #96: RFI: Is "Dual" Power harmful?

tinjaw, 2019-02-11:
I hookup my pISO to a cellphone charger to select the drives I want available and the drive I don't I turn off. Then I unplug the phone charger and plug the pISO into my PC. All is then glorious. All Hail pISO !!

Since I do it often, to reduce wear and tear on the hardware, I would like to leave it plugged into the cellphone charger when I plug it into the USB port of my PC. I would then like to be able to remove the pSIO from the PC and have it continued to be powered by the cellphone charger.

Can I do that? Should I NOT do that? What is the airspeed velocity of a Raspberry Pi laden swallow?

james-tate, 2019-02-11:
I would advise against this setup. While the setup should not be harmful, you are relying on the protection circuitry on the charger, and on your computer's USB port. The pi zero, or piso does not contain the necessary circuity. I don't think anything would happen to the pi or piso itself, but you are essentially plugging in your cell phone charger into your USB port on your computer.

tinjaw, 2019-02-11:
That's kinda what I suspected, but I've learned not to trust myself when money is involved. ;-) Thank you.

closed 2019-02-11 by tinjaw