ALSchwalm/pISO #100: seeking for your help about your fork

lionB777, 2021-03-06:
#Hi @No0ne,
I made one piso from your fork, but I don't have the key 2 working, just joystick up and down as expected, what check should I do to get it working ? thanks

No0ne, 2021-03-08:
https://github.com/No0ne/pISO/commit/dbb8a5f0d61b25107724497f50a708638e193ce2#diff-560156d3a7c437865b720d969a0fda0eebd224fb7df6fb1901c60b3b2c994e81

As you can see here, there are three buttons defined. The select key should be mapped to the button when you press the joystick in. Like on a Playstation controller the joystick is also a button.

lionB777, 2021-03-08:
ok, I missed that, because I made a custom case for it and the way it's made I have missed that. thanks for your help. I'm just a beginner swift coder so I can barely understand your code. Is there a chance that this push button and button 2 or other will work together? because definitely I'll not be able to code it myself. thanks

No0ne, 2021-03-08:
Do you use the Waveshare 1.3inch OLED HAT or something else?

lionB777, 2021-03-08:
I use the same as yours in the project.

lionB777, 2021-03-08:
[https://www.icloud.com/iclouddrive/0UXueoED_qb3Pm7_GWPdTgMPQ#image0](url)

No0ne, 2021-03-08:
Hardware approach: solder a wire from pin 33 to pin 38.
https://www.waveshare.com/w/upload/c/c8/1.3inch-OLED-HAL-Schematic.pdf

Software approach one button: change the line
`let select_input = Pin::new(13);`
to
`let select_input = Pin::new(20);`

Software approach both buttons: this would require more code changes!

lionB777, 2021-03-08:
is it possible to do something with a "or" condition ?

No0ne, 2021-03-08:
You need to do the initialization and listen for button presses separately for each pin.
If you don't want to code or compile the hardware approach with a wire between pin 33 and pin 38 on the gpio header or waveshare board should be the easiest.

lionB777, 2021-03-08:
the matter is not that I don't want to compile, it's more that I don't know how, even if if change the pin to 20 like you suggested in the source file

lionB777, 2021-03-08:
but willing to learn how if you don't mind to suggest a how to, please

No0ne, 2021-03-08:
Its documented here: https://github.com/ALSchwalm/pISO/blob/master/README.md

No0ne, 2021-03-08:
If you don't want to code or compile, the hardware approach with a wire between pin 33 and pin 38 on the gpio header or waveshare board should be the easiest solution.

lionB777, 2021-03-08:
ok I'll try, and I keep the hardware approach in mind, thank you.

lionB777, 2021-03-08:
last question, can the raspberry pi listen two buttons for the same action. 
so the code would be: 

       let select_input = Pin::new(13);
         select_input.export()?;
         select_input.set_direction(Direction::In)?;
         select_input.set_edge(Edge::FallingEdge)?;

        let select_input = Pin::new(20);
         select_input.export()?;
         select_input.set_direction(Direction::In)?;
         select_input.set_edge(Edge::FallingEdge)?;

lionB777, 2021-03-08:
coud it be something like this ?

 let select_input = Pin::new(13);
         select_input.export()?;
         select_input.set_direction(Direction::In)?;
         select_input.set_edge(Edge::FallingEdge)?;

let select_input2 = Pin::new(20);
         select_input.export()?;
         select_input.set_direction(Direction::In)?;
         select_input.set_edge(Edge::FallingEdge)?;

     ......
            down_input: down_input,
            select_input: select_input **(or, don't know the correct syntax)** select_input2,
            up_poller: up_poller,
            down_poller: down_poller,
            select_poller: select_poller,
            last_event: time::SystemTime::now(),