ALSchwalm/pISO #98: Green LED flashing 2 times after power on

Atheris142, 2021-03-06:
Hi, I was trying to make my own pISO with the github material. I ordered a custom pcb, solder the components and check it was right made. 

After following the steps of the wiki for flashing the image into the SD card the pISO does nothing, only the green LED flashes 2 times. I saw that LED code is when the pi can't read the SD card.

I have tried in another raspberry, and multiple SD cards and nothing changes. 

If someone can give me a hand i will appreciate a lot.

Thanks in advance.

bofhvp, 2021-11-21:
Hi
I've got the same problem too. Have you managed to resolve the problem, if so how did you go about it ?

Thanks

Atheris142, 2021-11-21:
Hi,
Unfortunately no. I was trying to flash the image a couple of times, watching the block size at formating the SD, trying multiple SD cards and nothing worked. 
I finally gave up and haven't tried anything else since.
I wish I could solve it, but I am not that experienced with these issues and the error seems quite vague.

bofhvp, 2021-11-22:
Hi

Which versions have you tried ? I've downloaded version both 1.2 and 1.3, but currently using 1.3 as its the "latest"

Atheris142, 2021-11-24:
I think I used the version 1.3 in all my attempts. I don't remember if I tried the 1.2 version.  

Maybe one of these days I will try it again, but I doubt I cuold make it work. Last time I followed step by step the process and didn't work. So I'm pretty lost.